goldshellminer/firmware #1: unstable hashrate CK5

abdulllh, 2021-06-12:
after latest update, hashrate is fluctuating between 10.2 - 13 TH/s

abdulllh, 2021-06-12:
[image]

hashrate before and after the update of CK5

Canonip, 2021-06-13:
It would be nice to have a repository with all firmware versions, so we can downgrade to the previous firmware in case there are any issues.

abdulllh, 2021-06-14:
> It would be nice to have a repository with all firmware versions, so we can downgrade to the previous firmware in case there are any issues.

I regret updating my miners 
I hope someone could make a custom firmware with higher hashrate

goldshellminer, 2021-06-16:
> after latest update, hashrate is fluctuating between 10.2 - 13 TH/s

> after latest update, hashrate is fluctuating between 10.2 - 13 TH/s

  the updates will not  affect hashrate in theory. How about the  hashrate in recent days?

closed 2022-09-05 by goldshellminer