goldshellminer/firmware #2: HS5 network connectivity

Genenex, 2021-06-14:
I just received my HS5 with the 2.0.3 firmware and .A3 hardware version.

The miner will boot up, give fast flashing green led (during startup procedure where fan goes from slow to fast and back to slow), then the miner gives a slow blinking red LED indicating some kind of network issue.

Tried various pools, reboots, resets, etc.  Network works, ethernet cable tested and working.  

Help??

goldshellminer, 2021-06-16:
> I just received my HS5 with the 2.0.3 firmware and .A3 hardware version.
>
> The miner will boot up, give fast flashing green led (during startup procedure where fan goes from slow to fast and back to [...]

Dear Customer:
           The miner  will ping  fixed server to check  network  is connected successful ot not .  please check if your firewall have disabled the ICMP

Genenex, 2021-06-16:
> Dear Customer:
> The miner will ping fixed server to check network is connected successful ot not . please check if your firewall have disabled the ICMP

Hi,

I have tried both with ICMP disabled and enabled, neither seems to fix the issue.  After changing to enabled from disabled I applied the settings, did a reboot of the router, then unplugged and waited 30 seconds and plugged back in.

Any other possible things I could try?

Genenex, 2021-06-23:
Got the HS5 to connect, tried 4 different routers.  One ASUS, two netgears, and one belkin.  The oldest netgear worked.

However, now the miner will run for about 16-20 hours then start disconnecting and restarting about 1-5 times an hour, causing extremely low hashrate and a lot of fan noise each time it restarts.
[image]

closed 2022-09-05 by goldshellminer