goldshellminer/firmware #3: KDBOX 2.0.7q Rejects issue

kwaichangdave, 2021-06-17:
After updating from 2.0.6q to 2.0.7q I now get 100% rejects from poolflare pool. Before it was under 1% rejects. I'm getting around 4% rejects on dxpool with 2.0.7q. I didn't test 2.0.6q on dxpool. How do I download 2.0.6q? It is now longer available. I want to revert back to 2.0.6q. And I tried rebooting and also setting back to factory settings.

kwaichangdave, 2021-06-18:
After 1.5 days on dxpool, my KDBOX is getting 5.3% rejects. Too high. My HS3SE has 0.1% reject rate after 1 month. Need a firmware update on HSBOX to fix the high reject rate.
[image]

AurelienMeray, 2021-06-18:
Experiencing the same issue on my KD box

kwaichangdave, 2021-06-18:
For the payment of $1,699 for one of these boxes, I hope they are working on a firmware fix. Do they even read this issue forum?

AurelienMeray, 2021-06-18:
> For the payment of $1,699 for one of these boxes, I hope they are working on a firmware fix. Do they even read this issue forum?

I hope so. I'd give it a few more days for a reply.

kwaichangdave, 2021-06-18:
Since Goldshell runs dxpool, they should count those rejected shares as paid shares.

dtb239, 2021-06-19:
Does anyone have 2.0.6q firmware file or 2.0.4 firmware file? The fans run too slow on 2.0.5 and 2.0.7q wont even work for me.

kwaichangdave, 2021-06-19:
> Does anyone have 2.0.6q firmware file or 2.0.4 firmware file? The fans run too slow on 2.0.5 and 2.0.7q wont even hash at full power for me.

Not available anymore. I want to have access to all the past firmwares. 2.0.7q is not working right.

dtb239, 2021-06-19:
@schua @goldshellminer please upload them ASAP i cannot operate on 2.0.5 temps too high and 2.0.7q causes all to be rejected in my pool. Please assist.

Canonip, 2021-06-21:
I'm linking my hotfix for this problem here: https://github.com/goldshellminer/firmware/issues/4#issuecomment-865359572

dubbb13, 2021-06-23:
it seems like goldshell is forcing us to use their pool... or else we would see 2.0.6 available for download

goldshellminer, 2021-06-27:
Please email to hello@goldshell.com or submit a ticket [submit](https://www.goldshell.com/ticket_new/)

tianchaijz, 2021-07-01:
Now poolflare fixed this issue, due to some legacy codes, the new 2.0.7 firmware add chain to the mining.submit params, doesn't pass the json schema validation.

closed 2022-09-05 by goldshellminer