goldshellminer/firmware #4: KD-Box Firmware Overheat 2.0.5 and Reject All on 2.0.7q

dtb239, 2021-06-19:
Please provide a previous firmware ASAP @goldshellminer @schua I'm unable to mine on multiple KD Box.

kwaichangdave, 2021-06-19:
Yes, please provide previous 2.0.4 and 2.0.6 for KDBOX.

Canonip, 2021-06-21:
my box was delivered with 2.0.6 - 
how the fuck do they only offer 2.0.5 and 2.0.7q for download.

how about offering every firmware version to download? jesus, you are acting like apple

kwaichangdave, 2021-06-21:
I have been emailing with John at goldshell support. Here is what he says about poolflare having all rejects with 2.0.7:
"Dear customer,
We have responded to your situation to the R & D department, and I believe that our 2.0.7 version will also support the pool you said soon.
Attached is version 2.0.6.
Kind regards.
Attachment(s)
firmware-Goldshell-KDBox-2.0.6.cpb"
He also said anything 5% and under for rejects is normal.

dtb239, 2021-06-21:
> I have been emailing with John at goldshell support. Here is what he says about poolflare having all rejects with 2.0.7:
> "Dear customer,
> We have responded to your situation to the R & D department, [...]

Can you please attach the file here.

kwaichangdave, 2021-06-21:
Github website is blocking the uploading of the 2.0.6 firmware cpb file.

Phara0h, 2021-06-22:
> Github website is blocking the uploading of the 2.0.6 firmware cpb file.
> I am also discussing this issue at Poolflare's discord. https://discord.com/channels/654674939550498846/701290331089534997

Can you just release the source code and then the community will help you fix these bugs and set up a proper release pipeline for free.

dubbb13, 2021-06-23:
i cant even get 2.0.5 to work at all. wont let me select miner tab..

Phara0h, 2021-06-23:
> i cant even get 2.0.5 to work at all. wont let me select miner tab..

mine was shipped with 2.0.4 guess I deff wont update until this becomes open source or better patch notes.

Canonip, 2021-06-23:
> wont let me select miner tab..

Try to disable addons like noscript or adblockers.
In my case uBlock Origin blocked the "Miner" tab

Phara0h, 2021-06-23:
> I have a hotfix for the old firmware, until Goldshell offers a working upgrade.
> 
> Got this Information by partly reverse engineering the firmware.
> 
> The miner has SSH access enabled. The password for user root is 19283746
> 
> When you are connected, use vi to edit product.json.
> Edit the temp_target value in the second last line.
> 
> ```
>         "fans": [
>             {
>                 "id":0,
>                 "name":"fan0",
>                 "rpm":40,
>                 "speed":"pwm0",
>                 "inipwm":50,
>                 "minpwm":20,
>                 "maxrpm":4500
>             },
>             {
>                 "id":1,
>                 "name":"fan1",
>                 "rpm":41,
>                 "speed":"pwm0",
>                 "inipwm":50,
>                 "minpwm":20,
>                 "maxrpm":4500
>             }
>         ],
>         "wifi":0,
>         "tempcutoff": 105.0,
>         "temp_target": 50.0
> }
> ```
> 
> I tested it with an (unrealistic) value of 50°C and the fans went on full blast (don't do this, use 80°C for example)
> 
> After that restart the miner from the webinterface
> 
> EDIT: do not try to modify other files and be careful, i am not responsible for you bricking your expensive device

Awesome work, here is the config on a 2.0.4 system for ref if anyone wants. Also if you find a way that I could dumb the firmware ill do it for you since 2.0.4 seems to run well.

```json
{
        "model":"Goldshell-KDBox",
        "cpbs": [
            {
                "id": 0,
                "reset": 67,
                "enable": 66,
                "power": {
                        "model" : "ISL8118",
                        "bus": "gpio",
                        "gpio0": 14,
                        "gpio1": 45,
                        "gpio2": 50,
                        "chan" : -1
                    },
                "tsensor" : {
                        "model": "tmp112",
                        "sensors": [
                            {
                                "bus": 0,
                                "addr": 72
                            },
                            {
                                "bus": 0,
                                "addr": 74
                            }
                        ]
                },
                "chips": {
                        "model": "ICA586",
                        "cnt": 16,
                        "layout": [2,2,2,2,2,2,2,2],
                        "power_mode" : "serial",
                        "serial_level": 8
                },
                "spichan": 1,
                "led": "cpb0",
                "plugin": 114,
                "spipath": "/dev/spidev1.0"
            }
        ],
        "fans": [
            {
                "id":0,
                "name":"fan0",
                "rpm":40,
                "speed":"pwm0",
                "inipwm":50,
                "minpwm":20,
                "maxrpm":4500
            },
            {
                "id":1,
                "name":"fan1",
                "rpm":41,
                "speed":"pwm0",
                "inipwm":50,
                "minpwm":20,
                "maxrpm":4500
            }
        ],
        "wifi":0,
        "tempcutoff": 105.0,
        "temp_target": 85.0
}
```

Phara0h, 2021-06-25:
> i cant even get 2.0.5 to work at all. wont let me select miner tab..

you using chrome on a modern computer?

trstaub, 2021-06-28:
> 
> 
> > wont let me select miner tab..
> 
> Try to disable addons like noscript or adblockers.
> In my case uBlock Origin blocked the "Miner" tab

Yes, I had the same issue on Firefox and uBlock.  I disabled uBlock for my local miner.  Good to know others are using that plug-in.