goldshellminer/firmware #6: Im pretty sure the KD-BOX is the same exact hardware as the DOGE-BOX

Phara0h, 2021-06-23:
Care to explain? if it is can we at least get the algorithms the hardware can run or open-source this so I can stop reverse engineering what I have here? Feel like I got kinda ripped off paying double for the same hardware but I will feel less bad if you open-source this firmware.

goldshellminer, 2021-06-27:
This firmware is not just an application. 
It's a whole platform and solution. The firmware will communicate with algo-chips directly.  

As we are still in a fiercely competitive environment (many algo-chip startup company established this year), we have no plan to publish right now.

Phara0h, 2021-06-29:
> This firmware is not just an application.
> It's a whole platform and solution. The firmware will communicate with algo-chips directly.
> 
> As we are still in a fiercely competitive environment (many algo-chip startup company established this year), we have no plan to publish right now.

Makes sense, don't want competitors to copy and paste code. You could make it GPL 3  and then sue if they don't make a contract with you. Do you guys use your own asics or use chips that all your competitors have access too?

closed 2022-09-15 by goldshellminer

luke-jr, 2026-09-05:
You are using my code, which is already GPL. You are REQUIRED to release it open source, or you are infringing!