goldshellminer/firmware #7: What is xminerd doing and why is it pinging www.baidu.com and www.wshifen.com?

Phara0h, 2021-06-23:
```bash
 [2021-06-24 06:01:40]  [Xminerd] NCF(0,1,0)
 [2021-06-24 06:01:43]  Start check net state...
 [2021-06-24 06:01:43]  PING www.wshifen.com IP 104.193.88.77
 [2021-06-24 06:01:43]  [Xminerd] ping www.baidu.com,Success
 [2021-06-24 06:01:43]  [Xminerd] fan0:(48, 2880)  fan1:(48, 2760)  . fan status: 0
 [2021-06-24 06:01:43]  [Xminerd] NCF(0,1,0)
 [2021-06-24 06:01:46]  Start check net state...
 [2021-06-24 06:01:46]  PING www.wshifen.com IP 104.193.88.123
 [2021-06-24 06:01:46]  [Xminerd] ping www.baidu.com,Success
 [2021-06-24 06:01:46]  [Xminerd] fan0:(48, 2820)  fan1:(48, 2760)  . fan status: 0
 [2021-06-24 06:01:46]  [Xminerd] NCF(0,1,0)
 [2021-06-24 06:01:49]  Start check net state...
 [2021-06-24 06:01:49]  PING www.wshifen.com IP 104.193.88.77
 [2021-06-24 06:01:49]  [Xminerd] ping www.baidu.com,Success
 [2021-06-24 06:01:49]  [Xminerd] fan0:(48, 2880)  fan1:(48, 2760)  . fan status: 0
 [2021-06-24 06:01:49]  [Xminerd] NCF(0,1,0)
 [2021-06-24 06:01:52]  Start check net state...
 [2021-06-24 06:01:52]  PING www.wshifen.com IP 104.193.88.123
 [2021-06-24 06:01:52]  [Xminerd] ping www.baidu.com,Success
 [2021-06-24 06:01:52]  [Xminerd] fan0:(48, 2880)  fan1:(48, 2760)  . fan status: 0
 [2021-06-24 06:01:52]  [Xminerd] NCF(0,1,0)
 [2021-06-24 06:01:55]  Start check net state...
 [2021-06-24 06:01:56]  PING www.wshifen.com IP 104.193.88.123
 [2021-06-24 06:01:56]  [Xminerd] ping www.baidu.com,Success
 [2021-06-24 06:01:56]  [Xminerd] fan0:(49, 2820)  fan1:(49, 2760)  . fan status: 0
 [2021-06-24 06:01:56]  [Xminerd] NCF(0,1,0)
 [2021-06-24 06:01:59]  Start check net state...
 [2021-06-24 06:01:59]  PING www.wshifen.com IP 104.193.88.77
 [2021-06-24 06:01:59]  [Xminerd] ping www.baidu.com,Success
 [2021-06-24 06:01:59]  [Xminerd] fan0:(49, 2880)  fan1:(49, 2760)  . fan status: 0
 [2021-06-24 06:01:59]  [Xminerd] NCF(0,1,0)
 [2021-06-24 06:02:02]  Start check net state...
 [2021-06-24 06:02:02]  PING www.wshifen.com IP 104.193.88.123
 [2021-06-24 06:02:02]  [Xminerd] ping www.baidu.com,Success
 [2021-06-24 06:02:02]  [Xminerd] fan0:(49, 2880)  fan1:(49, 2760)  . fan status: 0
 [2021-06-24 06:02:02]  [Xminerd] NCF(0,1,0)
 [2021-06-24 06:02:05]  Start check net state...
 [2021-06-24 06:02:05]  PING www.wshifen.com IP 104.193.88.123
 [2021-06-24 06:02:05]  [Xminerd] ping www.baidu.com,Success
 [2021-06-24 06:02:05]  [Xminerd] fan0:(49, 2880)  fan1:(49, 2820)  . fan status: 0
 [2021-06-24 06:02:05]  [Xminerd] NCF(0,1,0)
 [2021-06-24 06:02:08]  Start check net state...
 [2021-06-24 06:02:08]  PING www.wshifen.com IP 104.193.88.123
 [2021-06-24 06:02:08]  [Xminerd] ping www.baidu.com,Success
 [2021-06-24 06:02:08]  [Xminerd] fan0:(49, 2880)  fan1:(49, 2760)  . fan status: 0
 [2021-06-24 06:02:08]  [Xminerd] NCF(0,1,0)
 [2021-06-24 06:02:11]  Start check net state...
 [2021-06-24 06:02:11]  PING www.wshifen.com IP 104.193.88.123
 [2021-06-24 06:02:12]  [Xminerd] ping www.baidu.com,Success
 [2021-06-24 06:02:12]  [Xminerd] fan0:(49, 2880)  fan1:(49, 2760)  . fan status: 0
 [2021-06-24 06:02:12]  [Xminerd] NCF(0,1,0)
 [2021-06-24 06:02:15]  Start check net state...
 [2021-06-24 06:02:15]  PING www.wshifen.com IP 104.193.88.77
 [2021-06-24 06:02:15]  [Xminerd] ping www.baidu.com,Success
 [2021-06-24 06:02:15]  [Xminerd] fan0:(49, 2880)  fan1:(49, 2820)  . fan status: 0
 [2021-06-24 06:02:15]  [Xminerd] NCF(0,1,0)
```

Canonip, 2021-06-24:
Did you continue researching the miner :D

I **assume** it is just checking whether there is an internet connection.

btw there is a (pretty decent) debug page at http://minerip/#!/debug

Phara0h, 2021-06-25:
> Did you continue researching the miner :D
> 
> I **assume** it is just checking whether there is an internet connection.
> 
> btw there is a (pretty decent) debug page at http://minerip/#!/debug

I know that 2.0.5 and up added `www.wshifen.com` and have no idea why it would add another let another one to start with. Should be pinging 1.1.1.1 or 8.8.8.8

goldshellminer, 2021-06-27:
> I know that 2.0.5 and up added `www.wshifen.com` and have no idea why it would add another let another one to start with. Should be pinging 1.1.1.1 or 8.8.8.8

If you ping www.baidu.com, you will be redirect to ping www.wshifen.com. It is CNAME of baidu.com (largest search engine in China). Please check [https://www.google.com/search?q=shifen++baidu].

8.8.8.8 is google DNA sever.  It is not stable in China (GFW censorship).

We are testing other method to check the internet connection. 
We'll abandon the PING actions in later firmware release soon.

trstaub, 2021-06-28:
With 2.0.4 firmware:
Generally, I have observed the same ping activity to different domains.  It caught my attention when it was pinging Russia from the US shown on my Untangle firewall traffic map.  I have blocked it.  I wasn't sure why it was doing that.  And I did a DNS lookup, but not all of the IPs presented a domain name which caused alert.
I want to be covert/dark/quiet.  I don't want anyone to know I have a miner so I present less of a hack profile on the Internet.  For those who are security conscious, granted pings to a huge search engine may not be obvious, but pings to other specific IPs or domains can be viewed as a way of calling home.

Phara0h, 2021-06-29:
> > > Did you continue researching the miner :D
> > > I **assume** it is just checking whether there is an internet connection.
> > > btw there is a (pretty decent) debug page at http://minerip/#!/debug
> > 
> > 
> > I know that 2.0.5 and up added `www.wshifen.com` and have no idea why it would add another let another one to start with. Should be pinging 1.1.1.1 or 8.8.8.8
> 
> If you ping [www.baidu.com](http://www.baidu.com), you will be redirect to ping [www.wshifen.com](http://www.wshifen.com). It is CNAME of baidu.com (largest search engine in China). You can get more information here [google](https://www.google.com/search?q=shifen++baidu).
> 
> 8.8.8.8 is google DNA sever. It is not stable in China (GFW censorship).
> 
> We are testing other method to check the internet connection.
> We'll abandon the PING actions in later firmware release soon.

what about 1.1.1.1 I belive china allows that one?