goldshellminer/firmware #8: Afer the new firmware update CK-5 miner isn't mining and both red and green light are always on

GGamer162008, 2021-07-09:


GGamer162008, 2021-07-09:
Can someone tell me what is wrong.
[image]

Jwiscovitch, 2021-08-31:
I just downloaded the newest firmware and am having the same issue. I also tried to put the SD card in with the 2.0.3 firmware and nothing happened. I will sometimes get the red and green lights flashing in sync with each other. Still....not able to connect to the unit or anything. Has anyone found a fix?

MelasDigital, 2021-09-09:
I have the same problem with CK5. After the update, both LEDs are lit, the IP gets, but the web interface is unavailable.

Jwiscovitch, 2021-09-10:
I had to sd flash reset it. I did re-downloaded the new firmware and it works great now!

raminxxx, 2021-10-04:
> I had to sd flash reset it. I did re-downloaded the new firmware and it works great now!
how do you flash via sd card please explain i have this problem but don't know how to do that!

THECOMPANY2, 2021-11-10:
do somone hav the flash fill ASAP

zhaohaominer, 2021-12-25:
If you need to add my QQ notes GitHub, I can help

If you need to add my discord notes GitHub, I can help

I can provide burning files

QQ:86874708 Discord:zhaohao#5572

hakanhande, 2022-02-04:
Hey i need to this flash image i have same problem can you share with me 
my email   busarider@outlook.com 

zhaohaominer if you hear me can you share discord link again cuz its not working right now

M4773090, 2022-02-10:
Someone have IMG file to flash miner by SD?

closed 2022-09-15 by goldshellminer