goldshellminer/firmware #12: miniDoge 2.0.3 loses pool settings at random

djbadders, 2021-08-03:
Since upgrading to miniDoge v2.0.3 firmware the miners lose pool settings at random approximately once every 72 hours. The miner is still up but the pool settings are blank. A software based restart using the "System" menu does not resolve the issue, but a hard reboot by powering down restores the settings. Please can you resolve this ASAP.

goldshellminer, 2021-08-05:
This seems to be a problem with the control panel, please ask the customer service of Goldshell to deal with it.

djbadders, 2021-08-05:
I don't agree as rolling back to version 2.0.2 firmware has resolved the issue.

Jaseyy88, 2021-08-06:
I'm getting the exact same issue it just randomly goes offline every 3 days.

cubestring, 2021-08-07:
I have the same issue with 4 machines since firmware 2.03 update. 
They all go randomly offline and the Pool Settings under the Miner tab is blank.
The GUI restart will not resolve this but it requires a power down to restore and reconnect.
This happens on all 4 machines in a sequence you can count on for it to go offline again.
All network patch cords were replaced and network settings checked.
Where is it possible to receive firmware version 2.0.2 in order to roll back until the next version please?

m39869, 2021-08-08:
I have the same issue, one of the MiniDoge is on 2.0.0, and that's the most stable one. Also, I'm not sure why don't they share the release notes? There is no way to know what exactly is being fixed in a version. 

@cubestring, you can find 2.0.2 in the archive folder.

Runsun99, 2021-08-09:
Is that possible to downgrade firmware from 2.0.3 to 2.0.2 ?

Xandrianx, 2021-08-09:
I feel it necessary to tag in "me too". Incidentally I have observed that while the miner goes offline and no longer displays in the control panel - if I hard power cycle (unplug/replug) the pool settings show back up. Not sure if it is just me or not, but after reboot, the pool never goes active and it doesn't appear to do so until I change the DNS server I have specified to something else. It initiates a tcp/ip reset I believe and then poof - it's working again.

Had a bunch of this in syslog:
 [2021-08-08 22:49:28] Get Pool Name By Socket Error
 [2021-08-08 22:49:28] Get Pools Status Error
 [2021-08-08 22:49:28] Socket connect failed: Connection refused
 [2021-08-08 22:49:28] Exec {"command":"devs"} By Socket Error
 [2021-08-08 22:49:29] Socket connect failed: Connection refused
 [2021-08-08 22:49:29] Get Hshistory By Socket Error
 [2021-08-08 22:49:29] Socket connect failed: Connection refused
 [2021-08-08 22:49:29] Get Pool Name By Socket Error
 [2021-08-08 22:49:29] Get Pools Status Error

Miner syslog I think:
 [2021-08-08 10:40:44] MINIDOGE Miner detected, total 1 chains!
 [2021-08-08 10:40:44] Probing for an alive pool
 [2021-08-08 10:41:47] No servers were found that could be used to get work from.
 [2021-08-08 10:41:47] Please check the details from the list below of the servers you have input
 [2021-08-08 10:41:47] Most likely you have input the wrong URL, forgotten to add a port, or have not set up workers
 [2021-08-08 10:41:47] Pool: 0  URL: stratum+tcp://ltc.f2pool.com:8888  User: name  Password: pwd
 [2021-08-08 10:43:02] Started bfgminer 5.4.2
 [2021-08-08 10:43:03] Loaded configuration file /usr/config/bfgminer/bfgminer.json

djbadders, 2021-08-09:
> Is that possible to downgrade firmware from 2.0.3 to 2.0.2 ?

Yes I have done it successfully. I would recommend powering down and back up (i.e power switch not reset) before downgrading.

Xandrianx, 2021-08-10:
As a follow-up, I did a pre-emptive power cycle this everning, because I didn't want the unit to go offline while sleeping. Everything was hashing away just fine, pool was showing green/active. When it came up from boot - pool was inactive, not attempting to go active, just grey. Went to Debug log-monitor, saw a bunch of:
 monitor
  [2021-08-11 06:01:07]  [Xminerd] fan0:(50, 2880)  fan1:(50, 2880)  . fan status: 0
  [2021-08-11 06:01:07]  [Xminerd] NCF(0,1,0)
  [2021-08-11 06:01:07]  [Xminerd] Start Miner Reason NCF(0,1,0)!!
  [2021-08-11 06:01:10]  Start check net state...
  [2021-08-11 06:01:25]  [Xminerd] Couldn't resolve host "ltc.f2pool.com"
  [2021-08-11 06:01:25]  [Xminerd] any address 8888: Connection refused
  [2021-08-11 06:01:40]  [Xminerd] Couldn't resolve host "www.baidu.com"
  [2021-08-11 06:01:40]  any address 80 (www) open
  [2021-08-11 06:01:40]  [Xminerd] fan0:(50, 2880)  fan1:(50, 2880)  . fan status: 0
  [2021-08-11 06:01:40]  [Xminerd] NCF(0,1,0)
  [2021-08-11 06:01:43]  Start check net state...
  [2021-08-11 06:01:58]  [Xminerd] Couldn't resolve host "ltc.f2pool.com"
  [2021-08-11 06:01:58]  [Xminerd] any address 8888: Connection refused
  [2021-08-11 06:02:13]  [Xminerd] Couldn't resolve host "www.baidu.com"
  [2021-08-11 06:02:13]  any address 80 (www) open
  [2021-08-11 06:02:13]  [Xminerd] fan0:(50, 2880)  fan1:(50, 2820)  . fan status: 0
  [2021-08-11 06:02:13]  [Xminerd] NCF(0,1,0)
  [2021-08-11 06:02:13]  [Xminerd] Start Miner Reason NCF(0,1,0)!!
  [2021-08-11 06:02:16]  Start check net state...

All I did was add a 2nd DNS server, mostly to re-initiate the tcp/ip stack and things were back to normal. Miner/Pool Green.

onepulp, 2021-08-10:
I'm having the same issue on 2.0.3

MauiChris, 2021-08-13:
my 5 mini doge's, all on 2.0.3, seem to lock up ever 72hrs, forcing me to power cycle them. I haven't been losing pool settings after the power cycle though.

qweesy777, 2021-08-16:
> my 5 mini doge's, all on 2.0.3, seem to lock up ever 72hrs, forcing me to power cycle them. I haven't been losing pool settings after the power cycle though.

Same issue here

MauiChris, 2021-08-23:
looks like the firmware fixed this!!! finally made it to 4 days on all my miners without a crash :)

Hackpig1974, 2021-09-25:
> As a follow-up, I did a pre-emptive power cycle this everning, because I didn't want the unit to go offline while sleeping. Everything was hashing away just fine, pool was showing green/active. When [...]

This is still happening on the latest firmware version 2.1.2. They have never addressed this issue apparently. Quick tip: You don't have to change the DNS most times. I can literally hit APPLY and the miner can login to the pool shortly after. Reading through the issues mentioned here I don't think the developers are actively even monitoring the Issues section. @goldshellminer 

This is terrible because if the miners suffer a power loss, they will not reconnect when power is restored. This is unfortunate as ASICs are supposed to be bullet proof and mine with little to no interaction. Even my GPU miners will start up again after power loss. This is a simple bug in the code, very easy to fix.

bebraveboy, 2021-09-26:
So is the 2.0.0 firmware the temp. fix?

bebraveboy, 2021-09-26:
Or is it 2.0.2 firmware the ramp. fix?

Xandrianx, 2021-09-26:
Fix for static ip/dns? -- I put a dhcp assignment on my router to give the mini-doge the static IP I was manually setting on the box. After reboot, it does take it a little bit, but if I recall, it does eventually connect back up to the pool (2.1.1 firmware).  It is a "workaround" to having a 'static' ip of sorts.  The 2.1.1 firmware fixed the cause of mine and others mini-doges going sideways after 3days on the nose.

closed 2022-07-14 by goldshellminer