goldshellminer/firmware #13: KDA Box Miner Firmware Updates --Miners have vanished from the network and I can't access thru Goldshell

UNCTnutz82, 2021-08-17:
I just tried to update my KDA Box Miners(2) w/ firmware 2.0.9 and they are now off the network despite showing up on "Find My Miner".  I tried turning off/on and pushing the reset button, but the miners can't be accessed on the website anymore despite showing up there.  What do I do?  Do I need a new Driver (I'm using STMicroelectronics Virtual COM Port 3, 4, & 7) to match all the updated firmware for my KDA, Doge and other box miners?  I don't want to lose multiple days of revenue; so I am hesitant to update anymore of my miners.  Please assist if possible.  Thank you.

CAiston, 2021-08-17:
Some of my KD-Boxes change IP address after updating the firmware. Goldshell's Find My Miner does not update immediately, so even though it shows up there, the IP address it shows may be wrong. You should try software like Advanced IP Scanner or log in to your router admin to find your miner's current IP address. Once you find the IP address just type it into a browser from your home network to access the miner's dashboard. Also, go to your mining pool and see if your KD-Box is actively mining.

UNCTnutz82, 2021-08-17:
Strangely it is still mining.  I do think it’s related to an IP address change.

UNCTnutz82, 2021-08-17:
I think there’s still a problem with the firmware…I tried to do a default reset to factory settings but the firmware is still version 2.0.9
The IP addresses do keep changing but now my miners won’t come back online after adding the pool info.

PurebredTed, 2021-08-17:
I have observed on many KD-Boxes and on 2.0.4; 2.0.6; and 2.0.7q that each power cycle they generate a new MAC address - this is the root of the problem.  And if DHCP is used, that means they will also get a new IP Address.  DHCP is assigned to MAC addresses.  As such I can't assign static IP via DHCP because the MAC changes every time after a power cycle.  So, on the KD-Box, I disable DHCP and set them to a static IP address.  At least that way you can find them again after a power cycle.  You'll still see a new MAC address after every power cycle which is irritating.
I have created a case with Goldshell mentioning the same I typed above.  It seems they prefer a case over Github, but I think Github is better so we can share solutions until GoldShell has an updated Knowledgebase to share.  No doubt they are overloaded.

Additionally, after a power cycle, most of my KD-Boxes have red blinking lights so the network doesn't initialize properly after a power cycle.  I have to go in and nudge the network config to re-apply it which re-initilizes the network and gets things working again.  I have to reapply the DNS 1 server IP (in my case, I also use static DNS IP addresses).

The take-a-way from all this.  When the power goes out and I'm not home, they will all be offline until I have a way to manually nudge them back online again.  This is not good and I consider super critical and urgent to resolve.

UNCTnutz82, 2021-08-17:
Yeh this is very concerning. I think turning off DHCP might help in my case as well. I’m just not sure what I’m doing once I start assigning manual IP addresses; unless I just take the one it assigned last?  I can’t get my miners to boot back up now after a hard reset…goldshell told me to hard reset.  So now I’m wondering if I can go backward to the original firmware?? The reset function didn’t reset the firmware.

kwaichangdave, 2021-08-17:
Use the yottabc app provided at goldshell.com. Open the app and the push the ip report button on the miner. Instantly shows the miner's ip. Easy.

UNCTnutz82, 2021-08-17:
My miners will no longer mine no matter what I do…the firmware has really messed something up

UNCTnutz82, 2021-08-17:
I just tried to use yottabc, but I’ve never had any luck with it.  It doesn’t scan or show any miners.  I’m sure it’s something I’m doing wrong with my virus protection or firewall, but my Goldshell mining software is working with 3 smaller miners going. Any ideas??

UNCTnutz82, 2021-08-17:
I think I discovered a common theme.  Apparently poolflare is blocked now or incompatible with the new firmware. Sneaky update in my opinion if DXPool is their promoter pool.  Not sure though..maybe that’s why DXPool suddenly started to catch up in Hash Rate.

lucaspigato, 2021-11-06:
My KD5 not get hashrate, ridicolus.... this is software and pool issues cuz sometimes work fine and other not work.

closed 2022-09-15 by goldshellminer