goldshellminer/firmware #17: Leds of MineDOGE is always on when I update firmware to 2.1.1

WowWatchcate, 2021-09-02:
Old version firmware only blue led was flashing, MineDOGE works fine except the leds, thanks! 

[image]

[image]

MelasDigital, 2021-09-09:
I have the same problem with CK5. After the update, both LEDs are lit, the IP gets, but the web interface is unavailable.

goldshellminer, 2022-07-14:
> I have the same problem with CK5. After the update, both LEDs are lit, the IP gets, but the web interface is unavailable.

Have you tried factory reset yet? If not working, pls flash your card to the 2.2.3 firmware version.

closed 2022-07-14 by goldshellminer