goldshellminer/firmware #18: mini doge

payso112, 2021-09-09:
i need to update to something its not working i

Goldshell-MiniDOGE
ModelGoldshell-MiniDOGE

Firmware version2.1.1

Hardware version... 10.10.SA

Hackpig1974, 2021-09-25:
What is not working? You can't connect to a pool, you can't get an IP address? Can you be more specific?

payso112, 2021-09-25:
> What is not working? You can't connect to a pool, you can't get an IP address? Can you be more specific?

It's fixed now I'll delete the message...thanks tho

payso112, 2021-09-25:
It's fixed now .

DrunkenWarlock, 2021-11-10:
> Hey, My miner is not working, will not allow pool set up, no hashrate chart and continually errors with "Oops! Miner pool is not ready yet. Make sure miner is fully constructed."
> 
> Mini-Doge: FW - 2.1.3, HW - 10.10.SA
> 
> Can see it on my network, can toggle lights on/off, change passwords etc. but will not allow pools. It appears to be missing software or the firmware is bad. Does anyone know how to resolve or where I can download new software/FW 2.0.2?

I had a similar issue with my CK5 at one point and it seemed that my issue was from the pool I was using. I switched to a different pool and it started to work again. Initially I was using DxPool, then it went down and I switched over to F2Pool. Than DxPool made an update and I went back to it and it’s been working ever since. So I would try maybe switching up to a different pool and see if you get a different result

kwong2021, 2021-11-10:
Thanks, I tried both Dx and f2 with no luck. Miner just errors out when adding the pools. It will allow me to add and click apply, then errors as Unable to set up pools. I even tried setting the pools with Yotta BC, no luck, same error.

Hackpig1974, 2021-11-10:
More than likely you have something not right in the pool setup screen. The error you are getting is validated by the web server when you submit the changes. Can you post a screenshot or link of your setup for your pool? You can always blank or cross out the password if you like. I attached what mine looks like for reference.
[image]
.

kwong2021, 2021-11-10:
I added the same 3 in Yatta BC as you show with no luck (used my worker and PW), so I decided to use a different laptop (Chromebook) to access my miner and see if that would work. Went through the config steps again and now it appears to be working but drops out every 3-4 minutes then starts back up??? Strange, have you guys seen this with your mini's?

kwong2021, 2021-11-10:
[image]

kwong2021, 2021-11-10:
Thanks for all your help. I'll let it run for a while then send a ping if it continues to drop.

Hackpig1974, 2021-11-10:
The up and down on your graph is normal. Your hashrate fluctuates constantly on the miner.

Hackpig1974, 2021-11-10:
[image]

Here is one of my miners for reference.

kwong2021, 2021-11-10:
Yeah, mine started "loosing" the hashrate chart altogether for the fist 5-7 minutes but appears to be settling in now.

Hackpig1974, 2021-11-10:
Yeah, these are pretty set it and forget it. I typically don't open the web interface once they are setup. I check the pool daily to make sure my hashrate is at the level it normally is at, if I see it drop I go to: https://find.goldshell.com/ and then I connect to each miner real quick to see which one has a low hashrate, and why. I can do the same with the Yotta BC. Once configured if your network is stable you just wait for payouts and buy Lambo. ;-)

diaconom, 2021-11-10:
@Hackpig1974 you intentionally have your power plan set to balanced mode? You down by about 10Mhs

Hackpig1974, 2021-11-10:
> @Hackpig1974 you intentionally have your power plan set to balanced mode? You down by about 10Mhs

Yes, these are sitting around the house in the winter time. I don't have to run the furnace often for heat. But a couple of miners sit right next to the wife to keep here warm and they can't be too loud for conference calls. So I dial them back a bit for those 2 units close to her desk. I have my setup with a server power supply on a portable frame that allows me to move them around the house very easily. She points them at her when cold, and turns them away when they start to cook her. This setup has worked very well for me.  Let me know if you like it. 

[image]

kwong2021, 2021-11-10:
That sounds like a good plan. I have mine in my office and the wife uses a portable heated in her office. I think I'll move it downstairs and get ride of her portable. Thanks for the idea!

closed 2022-07-14 by goldshellminer