goldshellminer/firmware #19: powerplan or fan control missing

tojefest, 2021-09-12:
Hi, goldshell kd5 i can`t change anything. I have very cold place for this miner and fan works with 5100rpm. Is very big noise (noizy). Can i log into ssh or something to change that? I have temp 58C and cant change fan speed? oh come on..... With web Gui i saw only powerplan but is only one: hasratemode. Can anybody could help me? If there isn`t any control for poweruser is a huge issue.

jaspi72, 2021-09-17:
same on ck5.. 
I want to lower fan speed under 5100 if running temp is like now 70degrees.
And starting is wery problematic. 
If temp is low miner doesnt want to start so fan speed is way too hight on start.

jaspi72, 2021-09-17:
[image]

crypto-citizen, 2021-09-18:
Fan speeds were fine on CK5 with firmware 2.0.3, now with firmware 2.1.1 fans are constantly running @ 5100rpm. Same issue is with HS5. 
Goldshell, please enable us an option to change fan power mode based on ambient temperature🌡 or suggest alternative solution so that we can select fan speed via web interface. Thank you

crypto-citizen, 2021-09-18:
[image]

CK5

crypto-citizen, 2021-10-04:
Dear support, it's been two weeks since my last post. Can we get a reply on the above topic?

glob002, 2022-03-22:
Same to me, can you please add more options to tweak fan speed, I want to run it at 5600-5800 constantly, or at least add dropbox to pick % for fans

goldshellminer, 2022-07-14:
> `` Hi, goldshell kd5 i can`t change anything. I have very cold place for this miner and fan works with 5100rpm. Is very big noise (noizy). Can i log into ssh or something to change that? I have temp [...]

Hello, we have fixed this, pls upgrade your firmware to 2.2.3version.

goldshellminer, 2022-07-14:
> Dear support, it's been two weeks since my last post. Can we get a reply on the above topic?

Hello, we have fixed this, pls upgrade your CK5 firmware to 2.2.3version.

closed 2022-07-14 by goldshellminer

imannourih, 2024-03-13:
Hello, I have an lt5 pro miner.  I want to know how I can put my miner in low power mode, is it possible?