goldshellminer/firmware #20: Bruteforce password crack

Barzoul, 2021-09-12:
Hello,

Some users are facing bruteforce password hack on their goldshell miners when there is a port redirection. Please add the feature "after 3 attempts on password you must wait for 10 minutes" and after if possible "15 min... 3 attempts again... 30 min... etc....". It is a security base rule.

regards

Crypto1967, 2022-01-15:
I'm interested and don't know really what exactly you're talking about, but interested. I'm having network attacks from my house and it's a long story but it's a result of putting my ck5 and k1+ online, problems started. Anyway, My ck5 trouble free since 11/20 this week today and yesterday my 2miners mining bot notified me of being offline. Machine was hashing right on, yet somewhere other than my wallet. It kept changing the wallet and the mining pool. I would login see it, delete it, get mine back active, the problem kept doing the same thing. I messed it up trying to change fw I need an img file probably to restore it. Waiting on GS. I did change my PW to get into my ck5 as a security fix in my mind and I know what BF Attacks are.

PurebredTed, 2022-07-28:
Yes!  Upvote this!  Higher security is always better.

closed 2023-09-12 by goldshellminer