goldshellminer/firmware #23: DNS Resolution with Manual IP and DNS Assigned. Mine Doge Firmware 2.1.2

Hackpig1974, 2021-09-25:
I am seeing an issue with this firmware. (2.1.2 for Mini Doge) If you restart the device, when it boots back up it will not re-connect to the pool. It sits there not connected. You can make it connect by going to the system tab, and the clicking "Apply" under the DNS section. Then when you flip back to the miner tab you will see the pool connection indicator change from grey to green showing that is connected to the pool. Then the device will begin hashing again.

This may only be an issue when you have statically assigned an IP address and DNS. I can reproduce it every time by just rebooting the device. This should be investigated. Thanks. I have seen no other issues with this firmware.

cryptominerht, 2021-10-01:
I have the same problem - if i manually set IP address it said the mini doge is not connected to WAN but if i change it back to DHCP, it works again.

Hackpig1974, 2021-10-01:
@cryptominerht They really should investigate why static IP and DNS can't resolve properly on a reboot. It is so easy to reproduce and would me an easy fix at fixing an issue. So far, no response.

cryptominerht, 2021-10-01:
it is okay, I don't really see a different between 2.1.1 and 2.1.2, I have two  mini doges running on 2.1.1 and they are working just fine (and somehow on a slightly higher hash rate too). I may downgrade the new one to 2.1.1

Hackpig1974, 2021-10-01:
@cryptominerht I have 6 of them and on average have little to no issue with my mini doge. But if power flickers or there is an issue they don't resolve the pool. Maybe with enough time they will, not sure. If I get time I might just try and reboot one and start a timer to see how long it takes before it starts hashing again. I just hate playing with my money. HAHA

thomann061, 2022-01-07:
@goldshellminer Please fix this issue. When the miner is assigned a static ip address, and the power goes out or the machine reboots. We have to manually click apply on the DNS Settings page, to start the miner again.

goldshellminer, 2022-07-14:
> @cryptominerht I have 6 of them and on average have little to no issue with my mini doge. But if power flickers or there is an issue they don't resolve the pool. Maybe with enough time they will, [...]

Hello, have you met the same issue when upgrading to the recent firmware (2.1.3/2.1.4/2.2.3)?

goldshellminer, 2022-07-14:
> @goldshellminer Please fix this issue. When the miner is assigned a static ip address, and the power goes out or the machine reboots. We have to manually click apply on the DNS Settings page, to start the miner again.

Hello, have you met the same issue when upgrading to the recent firmware (2.1.3/2.1.4/2.2.3)?

thomann061, 2022-07-16:
Now, the static dns does not work....

goldshellminer, 2022-09-15:
> 

Pls contact the support team to get more support, Link: https://www.goldshell.com/support-center/

closed 2023-09-12 by goldshellminer