goldshellminer/firmware #34: Miner Hacked | Password keep changing | Brute Force?

ThefloorMiner, 2021-11-03:
I have an HS-BOX, by the time I changed the firmware to be able to mine Siacoin.
Whatever the size of the password. I'm hacked in less than one day.
So my thought it's been a security failure in the OS we downloaded.

Denisnalez, 2021-11-29:
It's possible to have extra security for the minner? Maybe an VPN? Maybe somebody entry every day or is something modified automatically inside the minner?

Denisnalez, 2021-11-30:
Finally I think I solved, I just changed mi router acces to MAC ID and put my devices one by one. Now teorecly nobody can entry in my ckbox.

goldshellminer, 2022-07-11:
> Finally I think I solved, I just changed mi router acces to MAC ID and put my devices one by one. Now teorecly nobody can entry in my ckbox.

> I have an HS-BOX, by the time I changed the firmware to be able to mine Siacoin. Whatever the size of the password. I'm hacked in less than one day. So my thought it's been a security failure in the OS we downloaded.

Hey, pls upgrade your miner to the latest 2.2.4 version, we improves system security: https://github.com/goldshellminer/firmware/blob/master/firmware-Goldshell-HSBox-2.2.4.cpb

closed 2022-07-11 by goldshellminer