goldshellminer/firmware #35: CKBOX unable to disable WiFi

gwaitsi, 2021-11-09:
I am unable to disable the WiFi on 

Firmware version2.1.1
Hardware version40.40.HA

The button is not selectable, or does not react to clicking on it.

gwaitsi, 2021-11-28:
this seems to be an initiation bug. Workaround is to connect to a wifi network, and then you are able to disable

closed 2022-07-11 by goldshellminer

plittlefield, 2023-12-19:
Did this work for you?

I have a KDBoxPro on firmware 2.2.6 and I cannot disable the WiFi ...

gwaitsi, 2023-12-19:
> Did this work for you?
> 
> I have a KDBoxPro on firmware 2.2.6 and I cannot disable the WiFi ...

did you try my suggestion? connect to a wifi network, then disable it.

plittlefield, 2023-12-19:
Just tried.  It connects to the wifi but **the toggle button is still inactive**.  This is my screen ...

[image]

plittlefield, 2023-12-19:
... so I pressed DISCONNECT and then tried the toggle slider but it is still inactive and does not work.