goldshellminer/firmware #38: Firmware KD-BOX

lbuijk, 2021-11-11:
Just did the latest firmware update on my KD-BOX (2.1.1.) and now it isn't booting - getting no IP. What to do?

EKotzin, 2021-11-23:
> Just did the latest firmware update on my KD-BOX (2.1.1.) and now it isn't booting - getting no IP. What to do?

I am having the exact same issuye.

rue182, 2021-11-28:
This applies to older KD box miners without WiFi connection as there are issues regarding the WiFi and the miners connection, that is a different issue. 

The below worked for me. My KD box just stopped mining one day, red light on by power plug and green light at top still blinking as well as ethernet blinking. Re-set the miner and it started normally, but then fans stayed on high and green light switched to red light by power again. Spent all day trying to figure this out. Then I ran into the thread in reddit below. when you add a pool in the goldshell, it will give you the preferred pools list buttons. I originally had Poolflare and no matter what I tried, the miner would not connect. Then I tried DxPool, same thing. I then followed the instruction in the below post and put them all in my goldshell miner pool settings and without assigning any one of them to mine from, it automatically selected F2Pool, the red light went to green, the fans quieted down, and hashing began on that pool. Make sure your miner is running when you set the pools up in Goldshell. Sign up for all of them and keep them in your pool list even if you don't plan on using most of them. Goldshell will automatically select a pool out of your list, like it did with mine, and start up normally again. I spent all day trying different things and the post below was the last thing I tried. It worked. Maybe there's a firmware issue and selecting your preferred pool is part of a larger issue, but until it gets fixed, at least your miner will be making money. I hope this helps:

Posted from a previous poster:

Yeah I posted this a while ago and re read what I posted. I don't think I did a very good job explaining my solution. SO 1. I'll try to re explain it better so it makes sense and 2. get you the f2pool settings. 1. Erasing all the preset pools from the miner and replacing them with preferred pools caused this for some reason. So even if you have Zero intention of visiting any of the 3. One of the 3 preset pools must remain or else you will have this problem. Good thing f2pool is part of the 3 original presets. Just leave the other 2 in place even if you're never going to mine there. This will prevent this problem.F2Pool Settings for litecoin: **<<<<<(GOLDSHELL FOR KD-BOX WILL AUTOFILL THESE FOR YOU WHEN YOU CLICK ON THEM - USER: will be your wallet that the payout goes to)**stratum+tcp://ltc.f2pool.com:8888stratum+tcp://ltc.f2pool.com:5200stratum+tcp://ltc.f2pool.com:3335

closed 2022-07-11 by goldshellminer