goldshellminer/firmware #40: HS-Box: Wifi does not connect to any nework

burgertyven, 2021-11-21:
-  HS-Box with wifi antenna
-  Purchased Nov 2021
-  firmware-Goldshell-HSBox-2.1.2.cpb
-  Does Wifi hardware recognise available networks: YES
-  Does Wifi connect to any networks: NO 

Tested:
- 2.4Ghz + 5Ghz
- DHCP + Manual IP assignment
- Confirmed that other devices connect to networks with success
- Downgrade of firmware to: firmware-Goldshell-HSBox-2.0.6q.cpb (does not support wifi)

burgertyven, 2021-12-10:
Solution: Goldshell Mini-Box miners dont support long Wifi passwords. Shorter passwords solved the issues.

closed 2021-12-10 by burgertyven

surfrider686, 2021-12-10:
I am not sure how this answers my question

Regards,
Ted Anderson
732-977-7743

burgertyven, 2022-01-23:
> I am not sure how this answers my question Regards, Ted Anderson 732-977-7743
> […](#)
> Sent from my iPhone

What ?