goldshellminer/firmware #43: KD Box no wireless adapter previously worked but after updating to firmware 2.1.1 Can't disable wifi

adcii1, 2021-11-30:
KD Box that has no wireless adapter previously worked but after updating to firmware 2.1.1, I can't disable wifi or get the unit back mining. I have tried dxpool.net, poolflare.net and f2pool, etc.. with the exact same issue. 

I have reset factory defaults, same issue.

I have assigned an ipv4 address & dns address same, issue.

Unit's causing my dns not to respond within my home network. 

United led light goes red near the network & power ports.

Issue's been ongoing for days.

Suspect more than one issue, pools & firmware/wireless settings creating conflict with a non wireless device.

Unit's worked previously with the exception of interment disconnects every 24hrs to 48hrs.

Please help

goldshellminer, 2022-07-11:
Have you solved this problem yet? if still not working, pls contact support for more solutions: https://www.goldshell.com/support-center/

closed 2022-07-11 by goldshellminer

PurebredTed, 2022-07-29:
I have 2 brand new KD-Box Pros on firmware 2.2.4
I can't turn off WiFi either and they are broadcasting an SSID which is unexpected for a client.
I haven't started to debug or look at logs yet.
I also just had to power cycle last night after its first week of mining.  All other older models are fine, just the new Pros have an issue.  One of the KD-Box Pros dropped off and stopped mining.  The other had sporadic hashrates.

KD-Box Pro Config:
Using ethernet | 2 Pools configured and using IP instead of URL and only difference is IP address | All Static IP setup | Cloud Control disabled