goldshellminer/firmware #50: KD Box Firmware 2.1.1 and 2.1.3 not Hashing

Sun3, 2022-01-03:
I just received two KD Boxes and one is working great, but the other one is not Hashing. The pool settings are the same for both boxes. I upgraded to firmware 2.1.3 but same results.

Screenshots
[https://ibb.co/sJMYzKc](https://ibb.co/sJMYzKc)
[https://ibb.co/pZxkd53](https://ibb.co/pZxkd53)

Any suggestions or help is appreciated.
Thank you.

diaconom, 2022-01-03:
Pic still shows 2.1.1 you sure firmware update worked ok?

I would power down the ‘working’ box and restart the other one after making sure using latest firmware and wait few mins and see if it starts to hash. If it does you would need to change the worker name just in case the pool is not allowing more than one worker with same name. 

My 2c

Sun3, 2022-01-04:
Sorry, the screenshot was taken before the update to 2.1.3. I did try to power down and I let it sit for a while. Then I plugged it back in with the same results. I even tried a few times to reset to factory settings, but the same problem, the machine is not hashing. I tried connecting to the dxpool without any worker name, the same result, but I do have two of the KD Boxes on dxpool working fine. I also tried Poolflare with the same results...

Not really sure what to do.

Sun3, 2022-01-05:
Goldshell said the KB Box chips are not starting and the box needs to be sent back to them. I also learned from Goldshell that by using the miner's IP address and add /debug you can get all sorts of information:

For example: http://10.0.0.115/#/debug

Hopefully this helps someone else with the same issue.

closed 2022-01-05 by Sun3