goldshellminer/firmware #54: Mini Doge Keeps shutting down

Dashinglight1945, 2022-01-25:
Goldshell-MiniDOGE
Firmware version 2.2.0
Hardware version 40.10.SA

Shuts down after an hour. How do I go back to version 2.1. or an earlier firmware version.

Alfz28, 2022-01-29:
i have the same issue, i had way old miner platform update yesterday. (v)2.1.2
                                                                                                                 (H) 10.10.sa

Alfz28, 2022-01-29:
hey dashinglight. dumb ??? how did you do a hardware update?

Dashinglight1945, 2022-01-29:
Yes. The issue is that the chips were overheating. The solution:
 Enter the goldShell dashboard.  (find.goldshell.com).  Click settings icon. Then you will see to the left "HOME, MINER, SYSTEM. 
Click the MINER tab. Then under "pool settings" is "Miner". Click the word miner 10 x consecutively. It will open the powerplan option. (mine looked like this 725 MHz 0.72 V 100 RPM 100 RPM). From there you can adjust/edit the hash rate MHZ and fan speed. RPM I needed to up my speed to 100%. Its working great now. 

To UPDATE FIRMWARE/hardware:
Click the SYSTEM tab. scroll down. under password you will see UPDATE. Click the attachment paperclip icon. and insert the downloaded updated from github for minidoge. Click UPDATE. Once the update is finished restart the miner.

Alfz28, 2022-01-29:
Thanks. I'll give it a try

Alfz28, 2022-01-30:
worked great. one question how high can you put the mh/z rate?? and thanks for the miner tip im running good now but more wondering how high you can take the mhz and hash out of control hahha. thanks if you have any info

Dashinglight1945, 2022-01-31:
I haven't found any stats for the mini doge specifically yet. But here is a stat for a goldshell CK5
Factory Setting : 850 MHZ 4.3 V 50 RPM 50 RPM
Overclock Set:    900 MHZ .55 V 60 RPM 60 RPM

Alfz28, 2022-01-31:
Ok, I'll check it out. Thanks again for your help miner is working perfect.
Thought it would be fun to see it hash out of control haha. I'll play with
those numbers. Thanks again

bkrigmo1, 2022-02-04:
Hi having this issue but it’s powering off just about every minute. Starts up and the fans ramp up then restarts and does it all over again. I tried all the firmware versions from latest to 2.0 nothing changes. I am now on latest version and have my setting under miner set to 725 MHz 0.72 V 100 RPM 100 RPM. Is this what you guys are set up with or did you modify? Can’t seem to get it to stabilize. My second miner is running perfectly fine with no issues at all and same settings except older firmware. (Already tried that version of firmware on the “broken” miner to no avail)

Dashinglight1945, 2022-02-04:
Lower your MHz to 600 and update your firmware to the newest version. Also, let your unit rest for a few hours and unplug it. Make sure it's not in an area that raises the temperature too much.

bkrigmo1, 2022-02-04:
I gave that a shot, but unfortunately, it still seems to be having the same
issues/behavior.

Dashinglight1945, 2022-02-04:
Send a message to support@pandamanshop.com They can help you look through your logs and find the root of the problem.

diaconom, 2022-02-04:
Is you device connected to hub.Goldshell.com ?

My device was failing (to hot) every few hrs after I upgraded to latest version, I noticed even if I set fans to 100 rpm the fans didn’t seem to be reving up. I tried changing power mode to balanced and that helped it would shutdown after 6hrs. 

I finally changed the power mode setting in the hub and as soon as I did that the fans revved up to 100% and slowly went down and the system has not rebooted yet.

It seems the settings in the Ui might not be passing on to the box but from the hub it seems to be working.

I suggest you try to change power mode on the hub and see if that helps keep the box alive?

bkrigmo1, 2022-02-05:
I have this a shot, but no luck. I can see that power mode is changing but
still shorting out every time it starts mining.

gwaitsi, 2022-05-17:
i got the same problem on a mini doge but only started after i upgraded to 2.21. my other units are fine though. strange.
my power plan is 725 Mhz 0.72 V 100 RPM 100 RPM

goldshellminer, 2022-07-11:
> i got the same problem on a mini doge but only started after i upgraded to 2.21. my other units are fine though. strange. my power plan is 725 Mhz 0.72 V 100 RPM 100 RPM

Hey, have you solved this problem yet? Pls upgrade to Mini-Doge 2.2.3 version and see if it is fixed. Link: https://github.com/goldshellminer/firmware/blob/master/firmware-Goldshell-MiniDOGE-2.2.3.cpb

closed 2022-07-11 by goldshellminer