goldshellminer/firmware #56: update kd5 now not working

stickygreen0, 2022-02-06:
made the update 2.2.1...  miner restart and then no acces and red and green light cant access miner

i need help

Dikrix75, 2022-03-08:
This is the same problem for me. Did you find a solution?

Technocityusa, 2022-03-08:
Mine was power unit . But try doing update firmware . If not , either control board or psu .

goldshellminer, 2022-07-11:
Hey, there are some solutions for you, pls follow the below steps:
1) Upgrade to the latest 2.2.3 version: https://github.com/goldshellminer/firmware/blob/master/firmware-Goldshell-KD5-2.2.3.cpb
2) If still not working, pls flash your SD card to check if solve or not. Contact support to get flash firmware: https://www.goldshell.com/support-center/

closed 2022-07-11 by goldshellminer