goldshellminer/firmware #57: STC box keep stopping

haiyatian, 2022-02-06:
Current Firmware version is 2.2.0

Symptoms: After few hours or days, miner will stop mining. Mining avg system show a few thousand KH/S but mining pool indicate miner is inactive. Currently each time miner stopped, I will restart it and it could work again.

Do anybody encounter similar problem?

diaconom, 2022-02-06:
Did you try updating firmware to 2.2.1 ?

haiyatian, 2022-02-08:
I had just updated to 2.2.1, issue remain.

goldshellminer, 2022-07-11:
Hey, pls download your miner log files and contact support for more solutions:
1) Download log files: https://twitter.com/goldshellminer/status/1541725472043610112?s=20&t=iClhAurPOAXKCVXA1tap9w
2) Contact support team: https://www.goldshell.com/support-center/