goldshellminer/firmware #63: Mini Doge start up issue

mshmity, 2022-02-26:
My mini doge worked fine for several months then started getting CPBO Failed to startup. I have loaded new firmware version 2.2.1 and even tried version 2.1.4 but same thing. I have heard older version of firmware work but can't find older versions.

Please help.

mshmity, 2022-02-27:
Ok now I have tried installing firmware version 2.1.1 from a micro SD card using the recommended steps. I can see that I have indeed installed firmware version 2.1.1 but still getting CPBO Failed to startup. 

Any Ideas????

goldshellminer, 2022-07-11:
This is probably that your hashboard CPB0 has an issue on it, pls contact support via the website link: https://www.goldshell.com/support-center/

closed 2022-07-11 by goldshellminer