goldshellminer/firmware #66: MiniDoge Firmware 2.2.1 FAULTY

JCBDigga, 2022-04-14:
Upgraded from 2.01 firmware due to having some issues with the miner loosing internet connection and seen recommendations to do so. However the firmware doesn't hash correctly maxed out at 22MH. Downgraded back to 2.01 and hit 160MH straight away again. 

In my experience, do not install this firmware.

diaconom, 2022-05-17:
> since i upgraded from 2.2.0 on one of my devices, it keeps getting thermal shutdown every couple of days. was fine for 6 months and just started within days of the upgrade

Reset to default settings - this resolved same issue I had with mine.

goldshellminer, 2022-07-11:
> 
Pls upgrade your firmware to the latest one 2.2.3version. We improve the fan control system: https://github.com/goldshellminer/firmware/blob/master/firmware-Goldshell-MiniDOGE-2.2.3.cpb.

closed 2022-07-11 by goldshellminer