goldshellminer/firmware #75: Kdlite

Cwolf9782, 2022-07-28:
Every couple day Kdlite stops hashing an loses pool info until it is powered cycled. Has firmware 2.2.7. Any one else haven issues

PurebredTed, 2022-07-29:
I have 2 brand new KD-Box Pros on firmware 2.2.4 that I just had to power cycle last night after its first week of mining.  All other older models are fine, just the new ones have an issue.  One of the KD-Box Pros dropped off and stopped mining.  The other had sporadic hashrates.
Can't turn off WiFi either and they are broadcasting a SSID which is unexpected for a client.
I haven't started to debug or look at logs yet.  Was just gonna start looking for new firmware.

Config:
Using ethernet | 2 Pools configured only difference is IP address | All Static IP setup | Cloud Control disabled

Cwolf9782, 2022-07-29:
I dnt have issues with my kd box’s just the Kdlite box haven issues I’ve talked with goldshell they said they will let their team know an should have firmware update in 2-3 wks

goldshellminer, 2022-08-05:
> I have 2 brand new KD-Box Pros on firmware 2.2.4 that I just had to power cycle last night after its first week of mining. All other older models are fine, just the new ones have an issue. One of [...]

Dear customer, we are working on this problem and getting a fix version. Pls contact the support team to get the relative firmware, link: https://goldshellhelp.zendesk.com/hc/en-us/requests/new?ticket_form_id=8017416782745

goldshellminer, 2022-08-05:
> I dnt have issues with my kd box’s just the Kdlite box haven issues I’ve talked with goldshell they said they will let their team know an should have firmware update in 2-3 wks Get Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

Dear customer, we are working on this problem and getting a fixed version. Pls contact the support team to get the relative firmware, link: https://goldshellhelp.zendesk.com/hc/en-us/requests/new?ticket_form_id=8017416782745

closed 2022-08-05 by goldshellminer