goldshellminer/firmware #76: KD6 Firmware update caused miner to stop hashing

KDMiner, 2022-07-29:
I updated my KD6 to firmware 2.2.10 and since then it no longer hashes.  Anyone have this issue or have previous versions of the firmware since Goldshell does not have it on their website?

LayLo3D, 2022-07-30:
**Firmware 2.2.7 -** https://goldshellhelp.zendesk.com/attachments/token/BtjOhLXJ97DI21KLL6eNnDZRs/?name=firmware-Goldshell-KD6-2.2.7.cpb

**Firmware 2.2.4 -** https://goldshellhelp.zendesk.com/attachments/token/DWAcPKY3t4pf002syuzZcYg9S/?name=firmware-Goldshell-KD6-2.2.4.cpb

I hope one of those will get you running again.

goldshellminer, 2022-08-05:
Dear customer, we are working on this problem and getting a fixed version. Pls contact the support team to get the relative firmware, link: https://goldshellhelp.zendesk.com/hc/en-us/requests/new?ticket_form_id=8017416782745

closed 2022-08-05 by goldshellminer