goldshellminer/firmware #101: Goldshell KA box pro issue with TH

vforv, 2024-06-09:
Instead of getting 1.6 TH, I am getting 1.2 on Ka box pro

[image]

What can be reason I updated firmware last avaliable Ka box

vforv, 2024-06-10:
I figured out that need some specific firmware for it can anyone send it?

jurado1993, 2024-06-10:
Donde podemos conseguir ese firmware

Spindler360, 2024-06-16:
This is also the case with my kaboxpro. 1.2 to 1.4 th after 2 days. strange

Spindler360, 2024-06-16:
btw @vforv I can see in your pic you have not the pro. the model discription shows "Goldshell-KABox" not "Goldshell-KABoxPro".
 
Have you perhaps installed the wrong update firmware? 

I see you use Firmware Version 2.2.7. This is only for the "normal" KABox. Not for the KaBoxPro.

closed 2024-06-16 by vforv

Pascalduc, 2024-11-07:
My KA Box Pro was working fine for a month then dropped to 1.3 TH/s.
I sent my log to Goldshell, but they want me to spend half the cost of the miner on courier shipping to send it back to Hong Kong.
With what I'm losing, the math doesn't make sense so I give them a bad review for poor after sale service.