goldshellminer/firmware #104: KA Box Pro High Temps

kansayvue, 2024-07-15:
Received my KA Box Pro today and seeing that it's hitting upwards of 80 degree in temps with fan running over 2500. just after 15 minutes. It fluctuate up and down on fan and temps but I've seen it hit over 80 in temps. This is similar to the last KA box I had where temp would rise and the fan would cycle upward to 2500 rpm. Adding a fan to the back doesn't help as it will cool down the box after a few minutes the fan rpm goes down then temp goes up. Any update on a new firmware like previous KA Box to minimize temp? I wouldn't mind the fan being at 2500 rpm.

bernaction, 2024-10-06:
We need to control more fans !!!!

Sebas212222, 2024-10-15:
Сделайте прошивку с низкой температурой ,устройство перегревается.Спасибо.